zapta/linbus #3: Analyzer not reading LIN messages

k3torres, 2015-02-23:
Hi Zapta,

First of all, thanks for all of your great work. Everything looks superb.

So I have fully soldered the Analyzer and checked over all components many times. Here is my outcome:

[image]

I was able to correctly program with Tiny ISP programmer. Analyzer has a heart beat and seems to work with your serial tool, but when I connect it to my LIN system in my Toyota Tundra, it does not read any LIN messages. I've tried adjusting baud rates to 19200, 9600, 4800, 2400, and also adjusting checksum to true/false for all. I am running out of ideas. Is there any tips of why I may not be getting any LIN messages?

-Kenny

zapta, 2015-02-23:
Hi Kenny,

Do you have access to an oscilloscope or logic analyzer? (to see what
signals are running on your car's lin bus)?

Also, where in the world are you located?

Zapta

k3torres, 2015-02-24:
I am located in Southern California, where are you?

Unfortunately, I do not have access to any logic analyzer or oscope. 

I have schematics diagrams for my Toyota and I am pretty sure I am accessing the correct LIN bus as well as ground and 12V power. Is it possible that Toyota uses an obscure baud rate? Or perhaps a rate that is faster than 19200?

zapta, 2015-02-24:
Do you know if the AT6631 generates any pulses on the LIN RX output? (you
can try connecting a LED in series with a resistor between the LIN RX and
ground).

You can also look for a ad hoc logic analyzer. It will help you identify
the speed and content of your linbus frames.

e.g. http://playground.arduino.cc/Main/LogicAnalyzer

http://www.ebay.com/itm/24MHz-8-Channel-USB-Logic-Analyzer-and-Decoder-Compatible-with-Saleae-Logic-/141580254471?pt=LH_DefaultDomain_0&hash=item20f6d73d07
   // The Saleae software support linbus decoding.

I am in northern California. I can mail you a loaner Microchip linbus
analyzer. It can generated linbus signals at arbitrary rate so you can test
your analyzer but it's not as useful as a logic analyzer that looks at the
actual signals in your car.

Zapta.

k3torres, 2015-02-24:
Great, I will test LIN RX with the LED. 

I will also purchase the Logic Analyzer from ebay.

That would be awesome about the loaner Microchip, or I could mail you my analyzer and you could test it against your LIN bus generator, which ever works best for you.

Thanks again for all your help. I will keep you updated.

zapta, 2015-02-24:
Hi Kenny,

You can mail it to me and I will test it here. Make sure to pack it well so
it doesn't get damaged.

Tal Dayan
14400 Blossom Hill
Los Gatos, CA 95032

As for the Saleae analyzer, I have the original 8 channel product but the
clones from ebay should also work (at least this is the Saleae guy said in
a recent Amp Hour episode). You can download and play with the software
now, even without the device. Last time I tried it the linbus decoder was
only in the beta version so if you don't see the linbus decoder try the
beta version.

I think that the logic analyzer does not support 12V so you will have to
connect it to the output of the linbus interface IC. Also beware of ground
loops, I used a laptop computer working on batteries.

Take care,

Tal

k3torres, 2015-02-27:
Hey Tal,

Just to give you an update, I ordered the Logic Analyzer and it should arrive by this weekend.

I also had an opportunity to test the LED against the RX and ground of the AT6631 and they revealed many pulses per second. So many pulses that the LED appears almost to stay lit consistently. I hope that this is good news. 

Maybe when the Logic Analyzer arrives I will have a better understanding, but what are you suggesting that I do with 12V output of my LIN component? 

Also, shall I still send the LIN bus Analyzer for testing?

Thanks again,

Kenny

zapta, 2015-02-27:
Hi Kenny,

If your logic analyzer will arrive this weekend take a look at the signal
at the AtA6631 RX output while connected to the linbus of your car. You
should pulses go from 5V (passive linbus state) to 0 volts (active linbus
state).  Let me know what you find or even send me the salee file (it
allows to save the capture data).

If it doesn't work or if you suspect that your analyzer doesn't work well
mail me the board and I will take a look.

I am not sure that I understand your question about the 12V.  A linbus has
three wires, GND, 12V and Linbus signal. You need to connect all the three
to the linbus analyzer. Otherwise it will not work.

Take care,

Tal

zapta, 2015-02-27:
... one more thing, the RX output of the ATA6631 is active low so you want
to connect the LED in serial with a resistor between the VCC (5V output of
the ATA6631) and the RXD. If the LED lights it means that there is LINBUS
activity.

Tal

k3torres, 2015-03-02:
Hi Tal,

So I have saved the capture data, where I have connected the Logic Analyzer to the LIN IC (RX and GND). To me the data looks obscure, and the Logic LIN analyzer function did not decode any LIN messages. This leads me to believe that my Analyzer is not properly solder or not working correctly. Maybe you can look at the capture data and come to some kind of better resolution than me. I can only paste images here on github "issues" section, can I email you the data (file extension .logicdata)? Here is a screen shot, sample rate is set to 8MHz for a 2 second recording. 

[image]

Also I did connect the LED to the IC using VCC and RX, and the LED stayed dimly lit, but was flickering. Another strange anomaly occurred when I setup this circuit, the Green LED "D6" on the Analyzer has been switching on and off rapidly as if it is receiving LIN messages, and even after I disconnected the series LED and resistor circuit the Green LED kept switching on and off. So I connected directly to the Analyzer via USB, and I used your serial tool. Then I started receiving LIN Error Messages when I did this and the Red LED on the Analyzer began turning on and off. Strange. LMK what you think.

Kenny

zapta, 2015-03-03:
Hi Kenny,

You can send me the saleae file to linbus@zapta.com. This way I will be
able to zoom in and see what data is coming.

Do you know for a fact that that bus is LINBUS?

k3torres, 2015-03-03:
Hey Tal,

I sent you the capture data by email as requested. To answer your question, I am pretty sure I am capturing data from a LINBUS in my truck. I have schematics that outline most of the electrical systems in my truck. They show me the color of wire and the label of meaning (LIN, IG+, GND).

zapta, 2015-03-03:
Hi Kenny, I looked at the data capture you sent and it doesn't look like
linbus signal at all. Do you want to mail me your board so I can test it?

Zapta

k3torres, 2015-03-03:
OMG TAL,

So I messed up on recording the capture data. The analyzer_bom.csv line 18 states a ATA6631 but the Digikey is ATA6663-TAQYCT-ND (what I ordered). So I was looking at the wrong IC pin configuration. I will retest with the correct pin configuration and send you back the data. Could this be a problem with the entire analyzer board? 

Kenny

k3torres, 2015-03-03:
Oh wow, I just looked at your picture on github of your analyzer board, clearly I have the wrong IC.

zapta, 2015-03-03:
Hi Kenny, yes, you are right, the BOM is wrong, should be 6631 instead of
6633. Sorry about that ;-)  I will update the BOM.

The 6631 is also supposed to provide the 5V for the MCU so I am not sure
what is happening with the 6633. Hopefully the MCU is still good (try to
measure where the +5V is supposed to be. If you get a lower voltage the MCU
is probably ok but am not sure).

I just checked my inventory and I have a few 6631's left. I can mailed you
one if you want.

Zapta

k3torres, 2015-03-04:
That's okay Tal, Im just glad we caught the problem. So are you asking me to measure where the VCC pin of ATA6631 is supposed to be?

Thanks for the kind gesture, but I already ordered the ATA6631 IC after I posted about it. It already has a ship status.

zapta, 2015-03-04:
My concern is that other ICs got damaged because the ATA6631 is supposed to
regulate the 12V to 5V. Yes, measure the voltage between pins 8 and 3 of
the ATA6633. If the voltage is below 5V then it probably didn't cause
additional damage.

k3torres, 2015-03-08:
Hi Tal,

So I got around to checking the voltage while connected to my LIN Bus and it turned out that the voltage was less than 5V. So I then replaced the old IC with the new and correct IC ATA6631. Reprogramed the MCU just in case. I then figured out that it doesn't use checksum and the baud rate is 9600. I'm reading lots of LIN messages now. Thanx so much for all the help. 

-Kenny

closed 2015-03-08 by k3torres

zapta, 2015-03-08:
That's good news!

If your goal is to identify how various switches and conditions are
reflected in linbus bits, I recommend using the python diff tool from
github. It report only changes in bits so it's more obvious what changes
while you play with the car's switches.

zapta, 2015-03-08:
... BTW, what is your end goal?

k3torres, 2015-03-08:
My plans for this project is to use the analyzer as a Master and take over the entire functions of the HVAC (Heater Ventilation and Air Conditioning) panel by receiving signals from an external source.

I hope this is possible. You did mention that the analyzer can be converted into a master in order to inject signals correct?

zapta, 2015-03-08:
All the linbus unit I built were not intended to replace the bus but to
work with it.  The Analyzer and Beeper monitor the bus silently and extract
signals. The Injector is actually a Linbus proxy that connects between the
master and a slave and flips certain bits of certain frames to inject
signals (kind of a 'man in the middle' attack).

Some terminology:
A linbus is made of 1 master and multiple slaves. The master functionality
doesn't send or receive any data, it just queries the slaves in some order
and each slave, when queried, responds with its data. The query is the
first part of the frame (which includes the slave ID byte) and the response
is the rest, including the checksum. That is, every frame includes a query
by slave ID from the master and an optional response from the slave
(sometimes the master queries slaves that do not exist). When a slave
responds with data, all the other slaves can read this data and act on it
if needed.

Typically the box that acts as a master, also need to send data. This is
done by implementing in the box, in addition to the master functionality
also slave(s) functionalities (each with its own unique id). That master
box can now send data by querying one of its own slaves which responds to
the query and all the other slaves on the bus can read that data and act on
it.

----

Once you will reverse engineer your bus (e.g. by hooking the analyzer,
toggling switches and see what bits are flipped, disconnecting boxes and
see what slave IDs stop responding to queries, etc), you will need to
decide what you want to do.

I am not sure for example that whatever you want to replace is actually the
bus master, you will need to figure this out. Also, if you replace the
master you will need to replace also all the slaves it implements.

My firmware doesn't implement a master or slave from scratch but the
hardware should be able to do that. Just remember that the linbus master
has an additional resistor and diode (R4 and D4 in this diagram
https://github.com/zapta/linbus/blob/master/injector/eagle/injector.pdf?raw=true
).

It's possible that the thing you want to replace is actually not the linbus
master but just a slave. The Analyzer will help you to explore your bus.

k3torres, 2015-03-08:
So this sounds disappointing. Are you saying I can't use the analyzer as a Master?

zapta, 2015-03-08:
The hardware can act as a master, assuming you will add that diode and
resistor. The software will require significant changes.

1. Are you sure that that box you want to replace is the actual linbus
master?

2. If it does, can you achieve your goal by leaving the master in place and
using an injector?