zapta/linbus #7: Linbus frame

alecsio, 2015-11-11:
Hi Zapta, 

I managed to capture few LIN frames from my Buick Regal 
 64 f2 fe 70 00 9d ERR
14 fc 70 90 ERR
14 fc 70 10 ERR
14 fe 70 90 ERR
64 72 ff 8e 00 fe ERR
64 f2 fe 70 00 99 ERR
14 fe 70 10 ERR
LIN errors: STOP SYNC

any ideea what's with the ERR ?

Thank you ,
Alex

zapta, 2015-11-12:
Hi Alex,

Not sure why the error. Need to look at the source code. Do you have access
to a logic analyzer and can capture the signals?

Is it possible that the baud rate or the linbus checksum version of your
car are different from the defaults in the analyzer?

Zapta

alecsio, 2015-11-12:
Do I need something special or a standard analizer will do ?
http://www.ebay.com/itm/24MHz-8-Channel-USB-Logic-Analyzer-and-Decoder-USB-FPGA-I2C-UART-SPI-AVR-/141662645942?hash=item20fbc06eb6:g:3lkAAOSwNSxVT3J9

zapta, 2015-11-12:
I think that the one in your link should be fine. Make sure to connect it
at the TTL level of the linbus signal, not on the linbus itself.

I used the original Saleae Logic 8 it's linbus analyzer worked great for
me. It's possible that the ebay one is a compatible clone.

tjb8907, 2016-08-03:
> Make sure to connect it
at the TTL level of the linbus signal, not on the linbus itself.

My logic analyzer should be here soon and I am going to start working on reading the frames that I need to replicate. I'm not sure what you mean by this though. Can you elaborate please?

zapta, 2016-08-04:
The signal on the linbus wires is 12 volts. Chances are it's too high for your logic analyzer. Connect your logic analyzer after the Linbus Atmel chip which reduces the 12V signals to 5V.

Connecting your logic analyzer directly to a 12V signal may damage it (check the specification of your logic analyzer what is the maximal input voltage that it can accept safely on its logic inputs)

tjb8907, 2016-08-04:
Ok that makes sense. I order one of the cheap eBay ones that you linked in another thread, guessing it is 5V limit. I don't have your analyzer board built yet, they just shipped from oshpark today, but I did get my order from Digikey so I do have the AT6631. 

Just a thought but is there a reason why a simple voltage divider wouldn't work? I'm guessing it would mess up the timing or something probably.

zapta, 2016-08-04:
>> Just a thought but is there a reason why a simple voltage divider wouldn't work? 

It may work. Too low resistors will overload the bus, too high will slow the signal.  Worth trying though.