zapta/linbus #8: Injector sources not compile

gioreva, 2015-12-15:
Arduino:1.7.8 (Windows 8.1), Scheda:"Arduino Uno"

lin_processor.cpp: In static member function 'static void lin_processor::StateReadData::handleIsr()':

lin_processor.cpp:555:67: error: no matching function for call to 'LinFrame::append_byte(uint8&)'

       rx_frame_buffers[head_frame_buffer].append_byte(byte_buffer_);

                                                                   ^

lin_processor.cpp:555:67: note: candidate is:

In file included from lin_processor.h:17:0,

                 from lin_processor.cpp:15:

lin_frame.h:65:15: note: void LinFrame::append_byte(uint8, boolean)

   inline void append_byte(uint8 value, boolean byte_has_injected_bits) {

               ^

lin_frame.h:65:15: note:   candidate expects 2 arguments, 1 provided

Errore durante la compilazione

  Questo report potrebbe essere più ricco
di informazioni con
  "Mostra un output dettagliato durante la compilazione"
  abilitato in "File > Impostazioni"

zapta, 2015-12-16:
Thanks for letting me know. I applied fix

d75af141fbb5d4a4b8bc0dcfc958bdd62bbb3ed1 which makes it to compile on
Arduino 1.6.5 / Mac OSX.  Where did you get the 1.7.8?  The last one I see
here is 1.6.6  https://www.arduino.cc/en/Main/Software

gioreva, 2015-12-16:
Hi Zapta

 

You found it on arduino.org

 

Why you don’t sell PCB of lin module ?

I was almost thinking of make and sell them, if possible.

A: zapta/linbus
Cc: gioreva
Oggetto: Re: [linbus] Injector sources not compile (#8)

zapta, 2015-12-16:
arduino.org are the bad guys, right?  ;-)

I am using the .cc IDE.

As for your question, I don't have the capacity to commercialize this
experimental design, nor do I have the expertise to comply with various
automative regulations and standards.

gioreva, 2015-12-21:
I'm studying to design a card with LIN.

Your Library (And Arduino Library) Work indifferently with UART 0, 1,2 and 3 ?

I could put  RS232, RS485, and LIN

The Rx/Tx switch of RS485 is expected by Adruino ?

zapta, 2015-12-22:
It doesn't use the UART but manipulate the incoming and the outgoing in
software (interrupts). This provide a lower delay (1 bit vs. 10 bits)
because it doesn't need to wait for the UART to read the entire byte before
sending it out.

To have the interrupt routing having predictable timing, any usage of
interrupts is carefully controlled. For that reason, the code is not
trivial. If all you need is LIN communication between Arduinos you can
probably have a simpler solution.

gioreva, 2015-12-24:
Email now work, not working before.

You had me say ?

 

Da: Giorgio Evangelista [mailto:g.evangelista@gevaelettronica.it] 
Inviato: lunedì 21 dicembre 2015 19:00
A: 'zapta/linbus'
Oggetto: R: [linbus] Injector sources not compile (#8)

 

I'm studying to design a card with LIN.

Your Library (And Arduino Library) Work indifferently with UART 0, 1,2 and 3 ?

I could put  RS232, RS485, and LIN

The Rx/Tx switch of RS485 is expected by Adruino ?

gioreva, 2015-12-26:
Hi Zapta.

I got the email failed. Now Work.

 

Can you help me design a card  TFT + CPU + LIN ?