zapta/linbus #10: Different LIN transceiver

slabiak, 2016-03-03:
Hello, I can't get the chip you used in your project (ATA6631) however i can get Philips TJA1020. Below you can see the example application note - my question is if there will be a lot of code change if i will use this chip instead of your chip, or is that chip is limited with some functions compared to your chip? 
[image]

zapta, 2016-03-04:
Hi,

Have you looked at the ATA6625 ?

Somebody suggested once that it may be a pin compatible replacement but I
never tried it myself.

Zapta

jkukiello, 2016-03-04:
I have tried the ATA6625C-TAQYCT-ND (from Digi-key) and it works fine and
is pin compatibile.
Best regards

Jacek Kukiełło

slabiak, 2016-03-05:
Thank you for your answers, it turned out that i have ATA662C in my home so i will use that chip, i have already did a circuit from the application circuit:
[image]
All i want to do is to do a LINBUS master which will send a frame which normally a control module sends to a car's lamp but i don't have that control module - i want to run that lamp without this module). I know the messege which control this lamp has ID 32 and the data is: "4C FF 32 52 05 00 7A FF" for LO lamp. I dont know how to send that information to the ATA662C so it will convert it to the LIN signal. Can you give me any advice on that? I read somewhere that all i need to do is to send these bytes one after another to serial TX? But it doesn't seem legit to me.

zapta, 2016-03-06:
My linbus firmware on github is implemented on the complex side with bit
banging in interrupts and such. The reason was that I wanted to have a
stack that can can inject signals to frames which was my ultimate goal.

If you just want to control a linbus slave you can probably do it easier
using standard serial UART. The only non standard part in the Linbus
communication is the break before each frame that typical UARTs cannot do
without some trickery.

I would search for other linbus projects that targets having a master (this
was not my goal). For example this https://github.com/gandrewstone/LIN

Your master need to do something along these lines:

1. Issue a break signal for the specified time. You can do it for example
by switching the UART output pin to be a standard digital output pin and
then set it LOW. When done, switch back to UART output. (or any other trick
you can think of).

2. Send the entire frame.

A few notes:
1. In linbus, the master doesn't send any data, it's just send the first
part of the frame with the id of the salve that need to respond. It's
basically going through a polling sequence, asking for responses from slave.

2. Only slaves send data, in response to the master calling their ID.

3. The box with the master can also include one or more slaves. In this
case, the same box sends both parts of the frame.

4. Each slave on the bus can see data from all the other salves. Sometimes
a slave use signals from multiple slaves for proper operation (e.g.
ignition state from one slave and button state from another slave).

Hope it helps,

Zapta.

jkukiello, 2016-03-06:
Well the answer is both yes and no...
Yes it's as simple as sending data to a tx buffer (if Your ATA is connected
to MCU TX)
and No as LIN expects You to send a synchronization break and
synchronization field.

Jacek Kukiełło

slabiak, 2016-03-06:
Hello, thanks for your answers - it helped me a lot. I used that library mentioned above and now I can send whatever frame I want, for example when i send  0x47, 0x20, 0xf8, 0x73, 0x00, 0x00, 0xcf, 0xff with frame ID 30 i get at LIN output this:
[image]
so it seems to be ok. However nothing happens when i send this frame. That's the thing I don't understand. As far as I know to turn on the ligjht I have to send some frame with ID 30 (for example) and some bytes. Which bulb will turn on depends on what bytes I will send. The thing I don't understand is the fact that you said that the master only sends a header (sync, id) and it is the slave who fill the bytes. So tell me how can I say the lamp: turn on LO lamp if I can't send any message to it? I thought that the messeage "turn on the LO lamp" is coded in bytes which master sends.

zapta, 2016-03-07:
I presume that you are using the correct baud rate and linbus checksum
version (there are two) for your device. Did you look at the linbus signals
in the car itself?

When you send the full frame, your box acts as both the master (sending
break, sync, id, etc) and the slave of ID 30 (sending the rest of the
frame). Hopefully the box you want to control (yet another slave with some
other id) read the data from the first slave and do whatever you want.

I know, it's confusing, only slave can control other slaves, master just
assign them frames that allow them to send.

The unit that you try to control doesn't necessary accepts 'commands' but
respond to bits it see on the bus from one or more other slave. Figuring
out what conditions will cause your unit to react is part of the trick of
reversing a linbus.

Hope it helps. Anyway, you had major progress being able to send full Lin
frames.

Zapta