zapta/linbus #16: Hardware

tjb8907, 2016-07-15:
I have been reading about your project and learning about Linbus, because i have a project that uses linbus to send commands, and I need to emulate those commands when certain criteria are met. These boards look like exactly what I am looking for. Do I have to have them made myself, or what is the easiest way to acquire one? I see that you have eagle files for them, but that is uncharted territory for me mostly. 

http://www.microchip.com/Developmenttools/ProductDetails.aspx?PartNO=APGDT001

would that be a similar product?

zapta, 2016-07-16:
Hi tjb,

The microchip device you mention should allow you to generate linbus frames and to look at linbus packets. However, I don't think it supports programatic API so you need to operate it through its app, on Windows only. 

I used it when I developing the linbus injector. It allowed me to simulate my car's environment in my lab.

The hardware designs in this repository should allow to generate linbus data but you will need to developed the firmware yourself, since the firmware here depends on an existing working linbus environment and they just monitor it or inject signals into packets.

Hope it helps,

Zapta

tjb8907, 2016-07-20:
Yep thats what I figured it would do. Just help with testing. Hook it up to help capture the frames that I want to duplicate. I might order it. 

What it the best way to go about getting the injector board made? Are there companies out there that will take your BOM and Schematic and make the PCB and assemble the components? I am capable of soldering some things, but SMD stuff I have no experience with. 

The project I am working on it basically identical to what you are doing, pushing a button automatically (based on another outside input). Writing the code and such won't be a problem. Especially if I can use Arduino.

tjb8907, 2016-07-27:
Updating this so I don't have to hijack the other thread with my project.

I ordered some injector boards and analyzer boards last night. Working on putting together an order for the components now.

My project with need an input to trigger the board to inject the message. I kind of had a revelation today that I should have made a small modification to the board that exposed a couple more pins of the 328, but then I kind of realized that once it is programmed, there would be no reason I couldn't use the ICSP Pins to connect a digital pin (D11 for example in Arduino world).  

I appreciate you giving me a little guidance. I will probably have a few questions when I start customizing the code!