zapta/linbus #17: Signal Injector

piripac, 2016-07-23:
Hi,  i have 2 gauge of a Smart roadster temp and turbo pressure, i want to install on a smart fortwo. The temp works, but the turbo pressure dont, the fortwo is diesel and the roadster is petrol, the pressure on the diesel is mesured on the manifold positive and the petrol is the vacum so  it is negative. Is it possible to read the value and change it? Do you Know how i can identify the value on the Smart lin protocol? Were can i buy your board?
Thanks.

zapta, 2016-07-23:
Hi priipac, 

If the pressure guage is connected via linbus than it may be to do the value mapping using this design, with some engineering work. First you need to know the format of the relevant packats on the original car, then the format on the new car, and then change the design of this circuit to do the mapping. Possibly you will also need to map packet ids, not just packet data.

I reversed engineered the linbus on my car using this same circuit which acts also as a packet sniffer, sending bus packets to my laptop computer. 

I don't know how familiar you are with electronics and firmware so can't comment how feasible it is.

BTW, I cannot provide boards or anything else. I just built this experimental circuit for my own car as a hobby and providing the info here for educational purposes. Maybe others here can help you.

Take care,

Zapta

piripac, 2016-07-24:
ok,  i just order 3 pcb and ask for samples at atmel, i gona try to make the board.
wish me luck.
thanks,
Nuno

zapta, 2016-07-25:
Hi Nuno,

Wish you good luck ;-)

You will need a programmer to program the firmware or at least the Arduino bootloader. This is done via the ISP port on the board. I am using this one and avrdude software on Mac OSX (my bash scripts are in this github repository)

http://www.atmel.com/tools/AVRISPMKII.aspx

There are many other programmers that will also work. If you are not familiar with it you can ask around.

Zapta.

tjb8907, 2016-07-26:
Hey Nuno,

where did you order the boards from?  There are all kinds of website, but not sure which one I should trust. I might have to learn how to solder SMD compenents....haha

zapta, 2016-07-26:
He probably ordered from oshpark (the hint is the number 3). It's a very good source in the US for small boards. You also want to check pcbshopper.com for comparison. 

As for soldering SMT, there are instructional videos on youtube, for example this one https://www.youtube.com/watch?v=5uiroWBkdFY  .  A tacky (gell) flus makes things much easier. I am using a cheap Chinese stereo microscope from ebay (a magnification of X3 to X10 is more than enough) but depending on your vision a magnifier glass or visor should be just fine.

Another option is to order from https://macrofab.com/ but then you need to do the rework if anything goes wrong. You can also get from ebay cheap SMT soldering training kits such as this one http://r.ebay.com/Rd0lzg

Good luck,

Zapta

piripac, 2016-07-26:
Hi,
I order from oshpark. And there are all kind  of vídeos on YouTube, maybe i can learn something

bootstrap18, 2016-07-26:
Hi, built this and wanted to a comment. The USB connector is hard to solder once the other parts are installed.  Do it early on. Normally one does low profile parts first but this is through-hole thinking.

I once had a lab partner who built a SMD board from a kit (solo) without knowing it was 'supposed' to be hard to do. He found it eaiser than through hole and got it done very quickly. Board worked first time. 

Regards,

PrzemekStankiewicz, 2016-08-11:
Hi, Were can i buy Linbus injector board?
Thanks.

Przemek

piripac, 2016-08-11:
Hi, i made mine on https://oshpark.com it eas very cheap to me 10usd with shipment to Portugal, 3 boards

piripac, 2017-11-13:
Hi, can I analyze Lin bus with new 991 board?

SparkyRih, 2017-12-13:
Hi,

I also own a Smart Roadster, and I bought some of the required hardware about a year ago (most of it I already have laying around), but never put it together due to lack of time, in about 3-4 months I might continue my project again...

The only issue I face at this moment is where is the LIN-BUS wiring on a Smart Roadster?

zapta, 2017-12-13:
I don't know the Smart Roadster and if it uses linbus but from my
experience, the linbus busses go to switch or LED clusters.

SparkyRih, 2017-12-13:
Well I was hoping @piripac would be able to answer this since he states he has this car...
I think I know where it is (violet wire behind the dashboard, very easily accessible from the passenger side), but I just want to be sure it is that wire before I start to do something with it...

piripac, 2017-12-13:
My project also on standby, on Sam unit pin14 on socket n11-9 it's violet, on for two don't have the wire, but the pin is usable

SparkyRih, 2017-12-13:
@piripac Can we contact via Facebook or different media? I'm working together with another guy on this project (he's also on hold though), another member joining us is always welcome :)

And before we mess up this nice GitHub page :)