zapta/linbus #21: Master as simple controller for windshield motor.

hugovw1976, 2017-05-02:
Hi zapta, I build a analyzer an successful read the mesages to controll a windshield motor from BMW.
I want to controll the motor out of BMW car, so only want to controll the motor with a buttons direct connected to the Atmega328p, I'm not a professional programmer and I wonder if you have a programm to direct write and send messages from a Lin master to a slave.
Thanks.

zapta, 2017-05-02:
Hi hugovw, what you need is a linbus master. I don't have code for one
since everything I did was done in the car and thus I used the car's master.

You can search for source code for linbus masters, maybe this one
https://github.com/John-Titor/LIN_Nodes  but most likely you will need to
do some programming to provide the linbus packets required by your motor.

Hope it helps.

hugovw1976, 2017-05-02:
Thanks, I will check.

hugovw1976, 2017-05-02:
In the injector code in wich file you send the code to activate the sport function?

zapta, 2017-05-02:
The injector code is not good as a master since it uses the car's master
and just modify packets as they go through.

The code that decides when to change the data packets is here
https://github.com/zapta/linbus/blob/master/injector/src_reference/arduino/custom_injector.h#L99
but I don't think it will be easy to convert it to a master.

One option you have is to buy this product
http://www.microchip.com/Developmenttools/ProductDetails.aspx?PartNO=APGDT001
 connect it to a compute and to your motor and send the packets that you
think will control your motor.

Z.

hugovw1976, 2017-05-02:
Ok, Thanks for your help.