zapta/linbus #22: OSH Park differs from source

clarkewd, 2017-06-07:
The Injector on OSH Park is labeled "linbus Injector Version C Rev1" and has this layout:

[image]

That's different from the pdfs in `injector/eagle` and `injector/eagle_991_mounted` - probably doesn't make much difference - just curious about the discrepancy.

zapta, 2017-06-08:
Hello,

I don't remember what the differences are but if you want, you can take a
look at the history of the .brd file in github and see what the differences
are.

I would think that the latest one is probably the best and you can upload
it to oshpark.

Hope it helped.

Zapta

amet, 2017-06-08:
Mounted has a hole for a screw and can be mounted onto a central console

[image]

zapta, 2017-06-08:
I also had success mounting PCBs with this double side tape.  Sometimes
directly (e.g. within a 3D printed case) and sometimes after wrapping them
with a wide shrink tube similar to RC's ESCs.

https://www.target.com/p/3m-extreme-mounting-tape-1-x60/-/A-14413487?ref=tgt_adv_XS000000&AFID=google_pla_df&CPNG=PLA_Home+Improvement+Shopping&adgroup=SC_Home+Improvement&LID=700000001170770pgs&network=g&device=c&location=9032142&gclid=CJG47KzrrtQCFQeTfgodZBIEmg&gclsrc=aw.ds