zapta/linbus #24: LIN bus rate

brunisgirl, 2017-12-27:
Hello,

I have tried your project on Arduino uno board, and it works perfectly fine. I have two problems:

1. When my LIN master sends a frame header (without data), and I print out the frame ID, I always get some garbage data on sio before the ID (which is detected correctly). What could be the reason for this?

2. When I send LIN messages fast (from an scheduler thread), I get SYNC error. What is the fastest that the code can receive new frame header after sending data (or better how can one measure this?)

Cheers,
S

zapta, 2017-12-28:
Hi brunisgirl,

Do you have access to an osciloscope or a logic analyzer?  It will allow
you to look at the signals.

Even a cheap 10-20$ one from amazon should be fine for linbus (which is
very slow).

And some of the, such as the saleae compatible, even come with a Linbus
decoder.

As for your project, I don't know enough details to comment. For example,
do you use a serial interface to generate the linbus master signals? For
example, how do you generate the sync period before each frame? Because
it's not a standard serial byte, I think it requires such trickery with the
serial interface and/or i/o port to generate.

Hope it helps,

Z.

brunisgirl, 2017-12-28:
Hi Z,

I do have an oscilloscope with LIN bus. But since my master is waiting for the response to send the new header, and the arduino goes to SYNC error, I can not send another frame meanwhile. However, the first header is received correctly and a response is generated from arduino (I have added a LIN send function to main loop after receive of a frame). I added the function just before the line:
 // Print frame to serial port.
 lin.send(frame.num_bytes());

void Lin::send(uint8_t nBytes)//(uint8_t addr, const uint8_t* message, uint8_t nBytes,uint8_t proto)
{
  byte message[] = {0x11, 0x22, 0x33, 0x44, 0x55, 0x66, 0x77, 0x88}; // LEDs ON
  //uint8_t addrbyte = (addr&0x3f) | addrParity(addr);
  uint8_t cksum = dataChecksum(message,8,0x32);
  serialBreak();       // Generate the low signal that exceeds 1 char.
  serial.write(0x55);  // Sync byte
  serial.write(0x32);  // ID byte
  serial.write(message, 8);  // data bytes
  serial.write(cksum);  // checksum
}

this function comes from: https://github.com/gandrewstone/LIN 
I have modified it a bit just for testing.
and the oscilloscope detects it as a LIN message.

For the master I use a Hercules board from TI, and with LIN transceiver in master mode. Then generating the sync or break is handled by the SoC on the board itself (i.e. I just send a header, the board adds the sync and break itself, which are detected by oscilloscope as well)

Best,
S

brunisgirl, 2018-01-17:
Hi,

I have figured the problem. It was the collision due to writing using the "sio" to the serial line as well.
Now, I would like to ask, how would you implement a slave response sending data? Does it make sense to add a new state after receiving the header where I send the response?

Thanks