zapta/linbus #27: Replacement of ATA6631/ATA6625

Yannik, 2019-01-24:
Both ATA6631 and ATA6625 have been deprecated by Microchip Technologies and can no longer be bought through the usual distributors. 

Is there any IC that can be used as a drop-in replacement?

Yannik, 2019-01-24:
Possibly the MCP2025? http://ww1.microchip.com/downloads/en/DeviceDoc/20002306B.pdf

AstronBnX, 2019-02-07:
You may use TJA1020 https://www.nxp.com/docs/en/data-sheet/TJA1020.pdf

Yannik, 2019-02-07:
@AstronBnX TJA1020 has a different pinout than ATA6631/ATA6625

AstronBnX, 2019-02-08:
@Yannik yes, you are right different pinouts but it is cheap. I'm also trying to get data from LINbus.

wchpikus, 2019-03-29:
And ata1020 working the same? Is it tested?
Second question, icm on volvo is master yes?

Yannik, 2019-03-29:
@wchpikus No, TJA1020 can not be used.

However, MCP2025 is a suitable replacement. I'm using MCP2025 and it works perfectly.

closed 2019-03-29 by Yannik

wchpikus, 2019-03-29:
i ordered ATA6631 for 1$ per pcs.
tnx