zapta/linbus #28: Write a frame

Thefirstalpha, 2019-02-12:
I've just discovered the LIN last week and I don't understand how can I write a frame using the injector part. Is it possible for a slave to write? Can someone help me?

Thanks

zapta, 2019-02-12:
The injector is not a slave. It sits on the bus between the master and
slave(s) and once in a while manipulate data bits passing through.

Writing a slave is very possible (depending on your coding skills) but a
different project.  ;-)

Thefirstalpha, 2019-02-13:
Ok, thanks for your response.
So if I understand, the injector read the Rx from the master bus, then it writes the same data on the Rx of the slave bus (or the opposite)? If that it's correct, can you show me which lines or files are doing the writing part on the other bus?

zapta, 2019-02-13:
In a linbus, only slaves send data. The send commands in the form 'slave X,
please send your data', each time to a different slave and that slave
response with its data which can be read by all other slaves connected to
the bus.

The injector is connected such that one slave is connected on one side and
the rest of the slaves are on the other side. This allows the injector to
1) read the data from all slaves and 2) change the data the one slave send
to the other slaves (it's a Man In The Middle attack).

I wouldn't use the injector code for writing a slave since it was designed
as a repeater, not something that generates data on its own, and it's code
is over complicated for a simple slave.