zapta/linbus #30: Enhanced checksum is used for diagnostic frames

Yannik, 2019-02-28:
Hello,
when setting `custom_defs::kUseLinChecksumVersion2` to true, the enhanced checksum is used for diagnostic frames too, leading to errors like this:

```
3c 7f 06 b2 23 17 46 01 03 43 ERR
7d 01 06 f2 01 00 00 00 ff 05 ERR
```

The LIN 2.x specification states that the classic checksum is to be used for diagnostic frames.
Source: https://assets.vector.com/cms/content/know-how/lin/LIN-Spec_2.2_Rev_A.PDF, page 20, point "classic checksum".

Yannik, 2019-03-01:
This is also described here:
https://www.lipowsky.com/assets/files/downloads/english/LINQuickStart_en.pdf

"These frames always have 8 data bytes and they always use the classic checksum."

zapta, 2019-03-01:
Thanks Yannik. I was not aware of those diagnostic packets.

More or less I designed it to work in my car and where happy when it did
(and still does).

Shouldn't be too difficult change the code to select the checksub method
based on the package type if anybody wants to do that.