zapta/linbus #32: Hardware and Software question

durante, 2019-03-17:
Hi Zapta, 

Great job with this project, just had a couple of questions regarding the project.
1. Do you sell the board completed? I think it could be a good buy for some people!

2. The project i am trying to put together is a LinBus controller, Where by the Lin Bus Analyser will be connected. It will read each signal and forward it on, 
With the ability of blocking set signals which can be activated / de-activated via a switch. 
Additionally, a SD Card shield will be connected as a data logger to help record then determine the desired signal to be blocked. 
I have been looking at the circuit, i believe we could use the SDA, SCL, A6, A7, D5 and D10  pins if we install the Arduino firmware.

[Injector.zip](https://github.com/zapta/linbus/files/2975737/Injector.zip)

I have attached the modified board, which we could develop a breakout board, to attach to suit a variety of more situations. Any advice or help..... especially with the code would be greatly appreciated.

Best regards

Durante

AstronBnX, 2019-03-19:
Hi Durante,

I absolutely wonder answer of your question 1. I need a completed LINBUS circuit board.

Thanks,
Reha

durante, 2019-03-20:
Hi Reha, Checked out at "Fusion PBC"  looks to expensive to order finished product ... roughly 50€ but naturally price comes down the more that are ordered! To make ourselves, cost would be about 18€ not including labor. 

Best option that i have found so far is from this company 

http://skpang.co.uk/catalog/linbus-breakout-board-p-1417.html

It is a shield / Breakout board, so for quick testing and usage. Funny enough, i think they copied all the work done by Zapta and made a simple version..... They even supply a link to Zapta`s work!

AstronBnX, 2019-03-20:
Hi again,

Thank you for information. 

I have been following this site for a long time, this product looks very expensive on this site. I agree with you about Zapta's work. Please have a look at this shield;

HW-A2 SunFlower Bus Shield Adapters Arduino; It is about 15USD. We know that they are using TJA1020/21 chip for linbus.
[https://bit.ly/2OeqKM3](https://bit.ly/2OeqKM3)
[image]

durante, 2019-03-20:
Hi Reha,

I did see this Shield, i was contemplating about getting one, but i am not a fan of it. I can not find any datasheet on it. Also i am unsure if it can run a Master Slave configuration like Zapta. Also i require 4 for my project, which makes problems with postage to Europe. 

Currently looking at building it on a breadboard, found another cheaper IC MCP2003A which has Master slave configuration. Then just require a few parts per set, 2 x 220pf, 1 x 10uf, 3 x 0.1uf, 1 x 22uf, 1 x 1K Ohm and 2 x MCP2003A. 1 x Aruino Total part cost around 6€ not including the Arduino. 

If the project goes well, i will further develope Zaptas board.

This way i have a little more freedom connecting extra leds, switches etc

AstronBnX, 2019-03-20:
Dear Durante,

Him, i have a customs problem in my country like yours. I think that better to make own board.. Just checked from Aliexpress MCP2003, and says one chip is 10Usd. https://bit.ly/2CsuG74

Which website did you see as lower price for MCP2003? I'm also interested in controlling or injecting Mercedes modules like seats buttons, air conditions buttons etc. 

I'll be please to you, if you let me know your developments. astronbnx at gmail.com

[image]

durante, 2019-03-21:
Hi Reha,

Ebay is our friend :-) 

https://www.ebay.de/itm/401624317217

4,28€ with delivery from England.

For the Capacitors, also from ebay.... but order a mixed packet brings the price down.

I have ordered mine, so probably start more next week when the parts arrive. My project is similar, main functions i need is seats and lighting, but on BMW. 

I will let you know on progress, but will take a while because my programming skills are not the best.

AstronBnX, 2019-03-21:
Hi Durante,

Yeah, Ebay is a Rock, but in my country Turkey it is forbidden for politics reasons :) I know that this is funny but it is real anyway.

Well, after you told me MCP2003, i have found it on the website arrow.com, then immediatelly ordered yesterday 10 pieces, they are on the way, thank you for notify me.

Do you use this circuit diagramme for your project? or different diagramme? If so, can you share with me?
[image]

I'm happy to hear that you are working on similar project. Please check my last project for iDrive Controller of BMW; [https://www.youtube.com/watch?v=Ida1PeciP7E&t=5s](https://www.youtube.com/watch?v=Ida1PeciP7E&t=5s)

 Please don't hesitate to contact with me about programming issue. I'm pretty fine for coding so far.

durante, 2019-03-23:
[image]

Hi, I really do not like politics, I believe it is only used to suppress the public. Good to hear you found another supplier.

Nice work on your project with the iDrive, have you done a write up yet and post?

I have supplied a simple schematic which i will be testing. The difficulty will be on the car. The LIN BUS wiring has to be disconnected from the initial master, connected to the Master side of the Arduino, Slave side gets connected to the rest of the vehicle. 

 Best setup would be to use Tyco Connectors so it can easily be connected/disconnected. 

Ideally, first program is to read and forward traffic. Also record to SD Card with time stamp.
Toggle switch for start stop, LED for indicating record, LED for SD Card Status, 

Second part once code is analyzed, to block set messages.

Zaptas board would keep it nice neat and tidy, but with that option not there at the moment. An Arduino allows just as good option.

zapta, 2019-03-23:
Hi Durante,

I don't do anything with this project. I just needed one for my car and it
is still running strong.

A company called T-Design is selling something similar product that they
adapted to certain car models. It may or may not work for your application.
(I am not affiliated with them, I just built one for my own use and put the
design in public domain for people that want to learn about Linbus)

https://t-design9.com/memory_module.html

Hope it helped.

Z.

ZeusEx1, 2024-04-02:
> [image]
>
> Hi, I really do not like politics, I believe it is only used to suppress the public. Good to hear you found another supplier.
>
> Nice work on your project with the iDrive, have you done a [...]

Hi Durante
Did you have any progress on your project?
Thanks

zapta, 2024-04-03:
Apropos recording to SD card. At the time I used a python script that got
the lin packets from the monitor board, processed them, and printed what
bit changed. This allowed me to identify  what change on the link bus when
I press a car button or a LED state is changed (e.g. sport mode vs normal
mode).

The script is here
https://github.com/zapta/linbus/blob/master/tools/serial/serial_dump.py .
Don't remember which board I used at that time to collect the packets,
maybe the one under the 'monitor' directory.