zapta/linbus #33: lin injector do not work at all

wchpikus, 2019-04-23:
Hi
This is my first post here, and i like to write what i like to do.
I lile to filter commands in volvo car (disable some function).
I build injector, but it not responses for lin frame.
I have olny one to 3 sek blink error message led.
Master connect to slave, slave to master.
Berret, i do not have any data on uart..
Could you tell me what can i check first?
Regards

wchpikus, 2019-04-23:
[image]

amet, 2019-04-24:
You are trying to inject Porsche LIN commands to Volvo, big chances that is not going to work.

You should use analyser to check the commands your slave is sending and then modify source to inject same commands to master

wchpikus, 2019-04-24:
Hehe..sure..
But, why any info i do not have on debug?
I saw in code massage like "waiting", but nothing send to uart..
Ok, i have analizer and can connect to rx pin behind lin transiver.
Do you have any example where i have to put lin frame (data) to analize?
I read about time to response, it us enough to analize and pass or not?
Tnx for response

amet, 2019-04-24:
There is a python script to help analyse the response from the board, try setting that up and check what the output is.