zapta/linbus #36: Obsolet parts

DrYouu, 2019-09-01:
I am not able to find lin bus transceiver
the one in the BOM file is obsolete in digikey and i dont know which ic could be compatible. 
Help please, i have already ordered pcbs.

robomaniac, 2019-09-02:
I search and they are all not pin compatible
https://www.digikey.ca/products/en/integrated-circuits-ics/interface-drivers-receivers-transceivers/710?k=&pkeyword=&sv=0&pv1291=46&sf=1&FV=1c0002%2C402d93%2C402d94%2C403180%2C401994%2C400009%2C402705%2C1140003%2C908001c%2C908016f%2C1f140000%2Cffe002c6%2C1a500002&quantity=&ColumnSort=0&page=1&pageSize=500

wort case you hand solder 8 wires per chip on the PCB.

Please check
https://github.com/zapta/linbus/issues/10

and

https://github.com/zapta/linbus/issues/27

some guys order directly from alibaba
https://www.alibaba.com/product-detail/-Electronic-Component-ATA6631_62122236739.html?spm=a2700.7724838.2017115.1.6b203e16BDGhNV

satbikkel, 2020-04-27:
MCP2025-500E/SN is the one i used..  worked perfect.