zapta/linbus #37: Is a prebuilt board available anywhere?

jbschooley, 2022-11-16:
Working on messing with my car's linbus and I need an injector. I don't even know where to start. From what I can see I'd have to order a board from OSH Park, buy all the components separately, and solder them on...I'm not confident in my ability to even buy the right components much less solder them on the board. So is there somewhere I can go to get this linbus injector prebuilt?

amet, 2022-12-11:
https://t-design9.com/memory_module_porsche.html

remulasce, 2023-05-15:
BTW, everything in this project works no problem on normal Arduinos with the same chip (ATMega 328). Tested both an Arduino Uno R3 and an Arduino Pro Mini using LINClick LIN transceivers. Likely any generic transceiver should work.

LiFePoRules, 2024-07-23:
@remulasce I currently try to rebuild like you described inyour comment, using an Arduino Uno and LINClick LIN transceivers. Can you provide some assistance in this process? I connected the boards (3.3V, GND, TX, RX), Uploaded the analyzer code (arduino.ino) and am wondering now, what are the next steps, or am I already ready to connect LIN to my LINshield? The output of the arduino IDE does not show my more concrete information other than the memory usage. Thank you for your help!

jbschooley, 2024-07-23:
I ended up designing and building a board for my project and it’s been working great for the last 6 months or so. Cost around $50 to order. I just need to find some time to publish a write up