zapta/linbus #38: Document that this project doesn't require custom hardware

remulasce, 2023-05-15:
It took a bit of reading and praying, but nothing in this codebase actually requires the custom circuit boards or firmware. It runs just fine on Arduinos with matching processors (ATMega 328). Tested both an Arduino Uno R3 and an Arduino Pro Mini. Any generic LIN transceiver can be used on the appropriate pins.

I have the pinouts as:
rx1:  D2 (receive from master via a 'slave' LIN transceiver)
tx1: D16

rx2: D15 (receive from your slave device via a 'master' LIN transceiver)
tx2: D4

Status LEDs on D7, D8, D9

Luke-K0, 2024-02-27:
> It took a bit of reading and praying, but nothing in this codebase actually requires the custom circuit boards or firmware. It runs just fine on Arduinos with matching processors (ATMega 328). [...]

Could you please share some more info about your setup? As far as I understand you connected two transcievers to one arduino, using pins mentioned above? Which transcievers did you use?

ciscospirit, 2025-01-03:
hey,
yah i would be also very interested how you solved it with a Arduino Uno Rev3
And which transcievers you have used and how you assambled it.
what software did you use than?

remulasce, 2025-02-23:
I used two of these: https://www.mikroe.com/lin-click
I cloned and modified this project to intercept a different set of LIN ids for a Tesla Model 3 rear drive unit oil pump to overclock it.

[image]